ChuckBuilds/LEDMatrix #14: Odds Scroller Displaying "No Odds Data"

leatherman5150, 2025-08-27:
Odds scroller was working fine until a couple of modifications to the scoreboard end, now constantly displaying "No odds data". Still switching back and forth to the scoreboard, which works great. 

Here's my code:
    "odds_ticker": {
        "enabled": true,
        "show_favorite_teams_only": false,
        "games_per_favorite_team": 1,
        "max_games_per_league": 16,
        "show_odds_only": true,
        "sort_order": "soonest",
        "enabled_leagues": ["nfl","mlb", "ncaa_fb"],
        "update_interval": 3600,
        "scroll_speed": 0.50,
        "scroll_delay": 0.01,
        "loop": true,
        "future_fetch_days": 7,
        "show_channel_logos": true

Thanks!

ChuckBuilds, 2025-08-27:
Hey there,
How long has this been happening? If it just started give it a day or so and see if it’s just the api limits. The odds ticker is a hog on api calls and it’s possible you hit the limit if you restarted the display a couple of times in a short period. It’s one of my more pressing issues to find a way to reduce calls or store them longer

leatherman5150, 2025-08-28:
Interesting... how does the display interpret API calls? Does it make a call every time a team's information is displayed? Is it time based? I'd assume the schedule scoreboard would call once a day to display next game data, but the scoreboard during a game would update once every few seconds?

closed 2025-09-15 by ChuckBuilds