ChuckBuilds/LEDMatrix #19: RPI Image w/ SSID Broadcast

ChuckBuilds, 2025-09-05:
**Describe the solution you'd like**
I want to create a baseline version of the LED Matrix that dramatically reduces the setup & barrier to entry. I would like to package the github project with requirements pre-installed directly to an RPI image that will broadcast it's SSID for easy wifi setup.

kmce2019, 2025-10-04:
I have a version that works. 
Normal image your pi
ssh into it and copy/paste the script

Reboot

On first boot the pi will broadcast PiSetup-XXXX (password raspberry), open http://10.42.0.1, submit your Wi-Fi.

After entering your Wi-Fi, the page says "Saved. Attempting to connect…"
<BUG> It continues to display this even after it has connected to your Wi-Fi

However, the pi is now on your network
<ENHANCEMENT> I'd like the webpage to say Connected, and tell the user the ip address to go to, or better yet, automatically redirect to it.

[make_ledmatrix_golden.sh](https://github.com/user-attachments/files/22690482/make_ledmatrix_golden.sh)

ChuckBuilds, 2026-09-16:
RPI image probably won't happen. Wifi Broadcast is working but could use refinement

closed 2026-09-16 by ChuckBuilds