ChuckBuilds/LEDMatrix #33: Some Managers don't properly handle multiple rows of panels

legoguy1000, 2025-09-10:
**Describe the bug**
When using multiple Rows in parallels, some of the managers are hard coded to a single panels height. So even if there are 2 rows, the height is determined using a single panels height.

```python
self.display_width = int(cols * chain)
self.display_height = hardware_config.get("rows", 32)
```

other managers and the correct method is to use the matrix's calculated height as shown

```python
self.display_width = self.display_manager.matrix.width
self.display_height = self.display_manager.matrix.height
```

affected managers:
- NBA
- NCAA FB
- NCAAM Basketball
- NFL
- NHL
- Soccer

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Use  full panel height accounting for multiple rows

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

legoguy1000, 2025-09-10:
I can create the MR later tonight. This started from https://github.com/ChuckBuilds/LEDMatrix/discussions/16#discussioncomment-14270725 and while I haven't gotten my panels yet, I am running this using https://github.com/ty-porter/RGBMatrixEmulator to play with it and the issue shows.

legoguy1000, 2025-09-10:
Also a few logo updates, NCAA FB, TEX is the ESPN abbreviation for UT Austin (Longhorns) which currently has the logo UT.png, it should be changed to TEX.png and the current logo using TEX.png is North Texas which for ESPN is UNT

ChuckBuilds, 2025-09-10:
Thanks for finding this. I can work it into my fixes on the dev branch. Just fixed those logo names in Dev.  Thanks for sharing that RGBMatrixEmulator as well - I need to check that out. It could really speed up my dev time.

legoguy1000, 2025-09-10:
Ya no problem. the emulator is really good, I'll probably submit a PR with a way to use it for dev.

legoguy1000, 2025-09-10:
> Thanks for finding this. I can work it into my fixes on the dev branch. Just fixed those logo names in Dev.  Thanks for sharing that RGBMatrixEmulator as well - I need to check that out. It could really speed up my dev time.
> 

Ya no problem. I've been looking for a project like this. I know there's at least one feed I want to create so I plan to dive in more. Once I'm up and running, happy to help imas much as I can

closed 2025-09-14 by legoguy1000