ChuckBuilds/LEDMatrix #37: Config is overwritten on update

legoguy1000, 2025-09-14:
**Describe the bug**
Because the config is tracked by Git, if u change the config and then try to update the code, it will error because of git conflicts. The config should be removed and a "example" config provided like the credentials. Then updates can be made. I'd also say that on service startup, it should read the config and if there are new configuration items, it should add the defaults to the original config and then it would always get updated.

ChuckBuilds, 2025-09-15:
That's a good idea for the config. Early on I had a lot more breaking changes through the config versions but maybe it's settled enough now.

ChuckBuilds, 2025-09-16:
got this working in dev I think

closed 2025-09-20 by legoguy1000