ChuckBuilds/LEDMatrix #42: for soccer section, how to add calls when special tournaments are on like world cup, euro etc

torontonightclub-oss, 2025-09-16:
for soccer section, how to add calls when special tournaments are on like world cup, euro etc.

torontonightclub-oss, 2025-09-16:
> I'll look into this. Right now it is only checking the ESPN Api if that
> team has a game scheduled. If they use a different end-point it would
> change. Would you like to only follow certain tournaments? I don't watch
> soccer so let me know what the intended use case is here

Yes just ...

World cup
Euro Cup
Concacaf
Club World Cup

These tournaments happen once every four years but every year one of these
are on.

http://www.torontonightclub.com

legoguy1000, 2025-10-06:
This is all the leagues provided by ESPN https://sports.core.api.espn.com/v2/sports/soccer/leagues, maybe during world cup years it will have more, not sure.