ChuckBuilds/LEDMatrix #74: NHL Manager not working correctly

bdk0172, 2025-09-27:
This is my NHL section in config.json:

    "nhl_scoreboard": {
        "enabled": true,
        "live_priority": true,
        "live_game_duration": 20,
        "show_odds": false,
        "test_mode": false,
        "update_interval_seconds": 3600,
        "live_update_interval": 30,
        "recent_update_interval": 3600,
        "upcoming_update_interval": 3600,
        "recent_games_to_show": 1,
        "upcoming_games_to_show": 1,
        "show_favorite_teams_only": false,
        "favorite_teams": [
            "MIN"
        ],
        "logo_dir": "assets/sports/nhl_logos",
        "show_records": true,
        "background_service": {
            "enabled": true,
            "max_workers": 3,
            "request_timeout": 30,
            "max_retries": 3,
            "priority": 2
        },
        "display_modes": {
            "nhl_live": true,
            "nhl_recent": true,
            "nhl_upcoming": true
        }
    },

From what's being logged, it appears it's getting data, but it's saying no events found.  If I put the URL in a browser it's returning a JSON response that looks legit.  Also odd that it's saying the ESPN API response when that's not where it's requesting the data from.  Was there a recent change from ESPN to nhle.com?

20:22:23.074 - INFO:src.nhl_managers:Fetching data from URL: https://api-web.nhle.com/v1/schedule/2025-09-26
20:22:23.551 - INFO:src.nhl_managers:[NHL] Successfully fetched data from NHL API for 2025-09-26
20:22:23.552 - INFO:src.nhl_managers:[NHL] Background data not available, fetching directly for nhl_20250927
20:22:23.553 - INFO:src.nhl_managers:Fetching data from URL: https://api-web.nhle.com/v1/schedule/2025-09-26
20:22:23.864 - INFO:src.nhl_managers:[NHL] Successfully fetched data from NHL API for 2025-09-26
20:22:23.942 - WARNING:src.nhl_managers:[NHL] No events found in ESPN API response
20:22:23.945 - INFO:src.nhl_managers:[NHL] Background data not available, fetching directly for nhl_20250927
20:22:23.946 - INFO:src.nhl_managers:[NHL] Using cached data for 2025-09-26
20:22:23.947 - WARNING:src.nhl_managers:[NHL] No events found in ESPN API response

bdk0172, 2025-09-27:
> I was messing with the data sources. Good catch. I thought the season
> hadn’t started yet so I didn’t test it. Looks like I missed preseason.
> I’ll
> try to fix this soon. (May not be this weekend) . Sorry! Thanks for the
> issue submission.

Wow!  Thanks for the quick response.  I was getting an error before the
issue I reported and dropped it into chatgpt to see what it thought.  Not
sure if this will help, but here's it's response.  I'm a developer, but
haven't worked with Python and don't have an environment set up to even try
and debug it.  I got my LED panels yesterday and got everything set up and
I just wanted to say you've done a great job with this.  I look forward to
your fix and will disable NHL until it's available.

in this github repository https://github.com/ChuckBuilds/LEDMatrix it's
getting an error: AttributeError: 'NHLRecentManager' object has no
attribute 'games_list' 18:02:50.115 - ERROR:src.display_controller:Error
during display update for mode nhl_recent: 'NHLRecentManager' object has no
attribute 'games_list' Traceback (most recent call last): File
"/home/ledpi/LEDMatrix/src/display_controller.py", line 1364, in run
manager_to_display.display(force_clear=self.force_clear) File
"/home/ledpi/LEDMatrix/src/nhl_managers.py", line 781, in display if not
self.games_list:
ChatGPT said:
Thought for 44s

Yep — that traceback means the class is trying to read self.games_list
before it’s ever created. In this repo the NHL “recent” display checks
games_list in its display() method (your log shows it hitting
nhl_managers.py:781), but the attribute isn’t set in the constructor, so
the very first call throws:

AttributeError: 'NHLRecentManager' object has no attribute 'games_list'

Quick, safe fix

Initialize the attribute once in the base NHL manager so every NHL manager
has it from the start:

# src/nhl_managers.py

 class BaseNHLManager:
     ...
     def __init__(self, config, display_manager, cache_manager):
         ...
         self.last_update = 0
         self.current_game = None+        # make sure subclasses can
read this before it's first populated+        self.games_list = []
         self.fonts = self._load_fonts()
         ...

And (defensive programming) make the check in NHLRecentManager.display()
tolerant even if a subclass forgets to initialize in the future:

-        if not self.games_list:+        if not getattr(self, "games_list", []):
             # draw “no recent games” or trigger a refresh

Either change on its own prevents the crash; doing both is best practice.
Why this helps

   -

   The display loop calls each manager’s display() before the first data
   refresh has necessarily run. If games_list isn’t pre-created, the first
   read explodes.
   -

   By initializing to [], the first frame just renders your “no recent
   games”/loading state instead of erroring; the next refresh populates the
   list.

Where to change it

   -

   BaseNHLManager.__init__ is in src/nhl_managers.py (you can see the NHL
   managers imported into the main controller here). GitHub
   <https://github.com/ChuckBuilds/LEDMatrix/raw/main/src/display_controller.py>

torontonightclub-oss, 2025-09-29:
has the update from today fixed the preseason games to start displaying because i just ran the update and its still not showing the preseason games, thanks!

closed 2025-10-06 by ChuckBuilds