ChuckBuilds/LEDMatrix #91: NHL Live Priority is not staying priority on the screen

ChuckBuilds, 2025-09-30:
**Describe the bug**
I have NHL Live Priority set to True and the Display is still rotating to other displays when the api data doesn't look live (?). It eventually rotated back into the game but I wonder if we need to figure out why this is happening if it continues to happen. 

**Expected behavior**
Stay in Live Priority game until the game is truly over. 

19:37:39.415 - INFO:src.display_controller:Rotating live priority sports: nhl -> nhl (duration: 30s)
19:38:09.010 - INFO:NHLLiveManager:Fetched 8 todays games for hockey - nhl
19:38:09.011 - INFO:src.odds_manager:Using cached odds from ESPN for odds_espn_hockey_nhl_401790331
19:38:09.364 - INFO:src.display_controller:Calling display method for nhl_live
19:38:09.431 - INFO:src.display_controller:Rotating live priority sports: nhl -> nhl (duration: 30s)
Weather data hasn't changed, using existing data
19:38:39.115 - INFO:NHLLiveManager:Fetched 8 todays games for hockey - nhl
19:38:39.117 - INFO:NHLLiveManager:No live/halftime games found for favorite teams.
19:38:39.117 - INFO:NHLLiveManager:Live games previously showing have ended or are no longer live.
19:38:39.119 - WARNING:NHLLiveManager:No game data available to display in NHLLiveManager
19:38:39.364 - INFO:src.display_controller:Calling display method for nhl_live
19:38:39.431 - INFO:src.display_controller:Switching to of_the_day from nhl_live
19:38:39.431 - INFO:src.display_controller:Showing of_the_day
19:38:39.431 - INFO:src.display_controller:manager_to_display is OfTheDayManager
19:38:39.431 - WARNING:src.of_the_day_manager:OfTheDayManager has no current items.
19:38:39.967 - INFO:MLB:Fetched 4 todays games for baseball - mlb
19:38:39.968 - INFO:MLB:No live/halftime games found for favorite teams.
19:39:10.107 - INFO:NFLLiveManager:Fetched 0 todays games for football - nfl
19:39:10.108 - INFO:NFLLiveManager:No live/halftime games found for favorite teams.
19:39:14.591 - INFO:NCAAFBLiveManager:Fetched 0 todays games for football - college-football
19:39:19.431 - DEBUG:src.of_the_day_manager:Internal rotation timer triggered, skipping external advance
19:39:19.431 - INFO:src.display_controller:Switching to nhl_recent from of_the_day
19:39:19.431 - INFO:src.display_controller:Showing nhl_recent
19:39:19.431 - INFO:src.display_controller:manager_to_display is NHLRecentManager
19:39:19.431 - INFO:src.display_controller:Calling display method for nhl_recent
19:39:49.481 - INFO:src.display_controller:Switching to nhl_upcoming from nhl_recent
19:39:49.481 - INFO:src.display_controller:Showing nhl_upcoming
19:39:49.481 - INFO:src.display_controller:manager_to_display is NHLUpcomingManager
19:39:49.481 - INFO:src.display_controller:Calling display method for nhl_upcoming
19:40:19.505 - INFO:src.display_controller:Switching to mlb_recent from nhl_upcoming
...
19:44:20.111 - INFO:NHLLiveManager:Fetched 8 todays games for hockey - nhl
19:44:20.112 - INFO:src.odds_manager:Using cached odds from ESPN for odds_espn_hockey_nhl_401790331
19:44:20.114 - INFO:NHLLiveManager:Found 1 live/halftime games for favorite teams.
19:44:20.114 - INFO:NHLLiveManager:  - FLA@TB (3:42 - 1st)
19:44:20.329 - INFO:MLB:Fetched 4 todays games for baseball - mlb
19:44:20.330 - INFO:MLB:No live/halftime games found for favorite teams.
19:44:20.531 - INFO:src.soccer_managers:[Soccer] Fetched data from ESPN API for usa.1 on 20250929
19:44:20.728 - INFO:src.soccer_managers:[Soccer] Fetched data from ESPN API for usa.1 on 20250930
19:44:20.923 - INFO:src.soccer_managers:[Soccer] Fetched data from ESPN API for usa.1 on 20251001
19:44:20.924 - INFO:src.soccer_managers:[Soccer] No live games found matching favorite teams.
19:44:20.924 - INFO:src.display_controller:Live priority takeover: Switching to nhl_live from clock
19:44:20.924 - INFO:src.display_controller:Showing nhl_live
19:44:20.924 - INFO:src.display_controller:manager_to_display is NHLLiveManager
19:44:20.924 - INFO:src.display_controller:Calling display method for nhl_live
19:44:50.022 - INFO:NHLLiveManager:Fetched 8 todays games for hockey - nhl
19:44:50.022 - INFO:src.odds_manager:Using cached odds from ESPN for odds_espn_hockey_nhl_401790331
19:44:50.025 - INFO:src.display_controller:Rotating live priority sports: nhl -> nhl (duration: 30s)
19:44:50.956 - INFO:src.display_controller:Calling display method for nhl_live

ChuckBuilds, 2025-10-01:
happened again, just documenting: 
20:04:21.481 - INFO:src.display_controller:Rotating live priority sports: nhl -> nhl (duration: 30s)
20:04:22.248 - INFO:src.display_controller:Calling display method for nhl_live
20:04:51.489 - INFO:NHLLiveManager:Fetched 8 todays games for hockey - nhl
20:04:51.490 - INFO:src.odds_manager:Using cached odds from ESPN for odds_espn_hockey_nhl_401790331
20:04:51.492 - INFO:src.display_controller:Rotating live priority sports: nhl -> nhl (duration: 30s)
20:04:52.264 - INFO:src.display_controller:Calling display method for nhl_live
20:05:21.805 - INFO:NHLLiveManager:Fetched 8 todays games for hockey - nhl
20:05:21.807 - INFO:NHLLiveManager:No live/halftime games found for favorite teams.
20:05:21.807 - INFO:NHLLiveManager:Live games previously showing have ended or are no longer live.
20:05:21.809 - INFO:src.display_controller:Switching to weather_current from nhl_live
20:05:21.809 - INFO:src.display_controller:Showing weather_current
20:05:21.809 - INFO:src.display_controller:manager_to_display is WeatherManager

legoguy1000, 2025-10-01:
I know what's up, the time between periods, the status is different like halftime so is dripping ur as live until it starts up again

legoguy1000, 2025-10-01:
I think something else is weird cause I haven't rotated out of live view once, been rotating between live MLB and NHL and in my emulator, all the live NHL games but no non-live games. I didn't find a few other bugs though. see #93

torontonightclub-oss, 2025-10-02:
im using the latest update have these nhl changes been updated to the server?  I still get nothing to display when NHL is set to true and with favourite teams only checked and unchecked.  nothing displays for NHL.  it does show up in the ticker tho.  just not for its alotted slot in the cycle like the other sports.

legoguy1000, 2025-10-02:
Can u share ur config? Yes nhl code is in main.

torontonightclub-oss, 2025-10-02:
"display_durations": {
            "clock": 15,
            "weather": 10,
            "stocks": 30,
            "hourly_forecast": 10,
            "daily_forecast": 10,
            "stock_news": 20,
            "odds_ticker": 140,
            "leaderboard": 60,
            "nhl_live": 30,
            "nhl_recent": 30,
            "nhl_upcoming": 30,
            "nba_live": 30,
            "nba_recent": 30,
            "nba_upcoming": 30,
            "nfl_live": 30,
            "nfl_recent": 50,
            "nfl_upcoming": 50,
            "ncaa_fb_live": 30,
            "ncaa_fb_recent": 30,

    "nhl_scoreboard": {
        "enabled": true,
        "live_priority": false,
        "live_game_duration": 60,
        "show_odds": false,
        "test_mode": false,
        "update_interval_seconds": 3600,
        "live_update_interval": 30,
        "recent_update_interval": 3600,
        "upcoming_update_interval": 3600,
        "recent_games_to_show": 1,
        "upcoming_games_to_show": 2,
        "show_favorite_teams_only": false,
        "favorite_teams": [
            "TOR"
        ],
        "logo_dir": "assets/sports/nhl_logos",
        "show_records": true,
        "background_service": {
            "enabled": true,
            "max_workers": 3,
            "request_timeout": 30,
            "max_retries": 3,
            "priority": 2
        },
        "display_modes": {
            "nhl_live": true,
            "nhl_recent": true,
            "nhl_upcoming": true
        }
    },

legoguy1000, 2025-10-02:
Do u see any NHL logs in the output? Have u restarted the service since u changed the config?

torontonightclub-oss, 2025-10-02:
No restart the service but I rebooted several times

http://www.torontonightclub.com

torontonightclub-oss, 2025-10-02:
Oct 01 21:10:04 ledpi ledmatrix[480]: 21:10:04.038 - INFO:src.nhl_managers:Fetching data from URL: https://api-web.nhle.com/v1/schedule/2025-10-01
Oct 01 21:10:05 ledpi ledmatrix[480]: 21:10:05.174 - INFO:src.nhl_managers:[NHL] Successfully fetched data from NHL API for 2025-10-01
Oct 01 21:10:05 ledpi ledmatrix[480]: 21:10:05.194 - INFO:src.nhl_managers:Fetching data from URL: https://api-web.nhle.com/v1/schedule/2025-10-01
Oct 01 21:10:06 ledpi ledmatrix[480]:

torontonightclub-oss, 2025-10-02:
1:10:06 ledpi ledmatrix[480]: 21:10:06.352 - WARNING:src.nhl_managers:[NHL] No events found in ESPN API response
Oct 01 21:10:06 ledpi ledmatrix[480]: 21:10:06.354 - INFO:src.nhl_managers:[NHL] Using cached data for 2025-10-01
Oct 01 21:10:06 ledpi ledmatrix[480]: 21:10:06.355 - WARNING:src.nhl_managers:[NHL] No events found in ESPN API response

legoguy1000, 2025-10-02:
> Oct 01 21:10:04 ledpi ledmatrix[480]: 21:10:04.038 - INFO:src.nhl_managers:Fetching data from URL: https://api-web.nhle.com/v1/schedule/2025-10-01
> Oct 01 21:10:05 ledpi ledmatrix[480]: 21:10:05.174 [...]

This is out of date. The code that was merged to main only uses ESPN data, not able. I would make sure the comit ur on is the latest.

torontonightclub-oss, 2025-10-02:
im getting alot of this when i enter that command

Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.416 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.417 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.418 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.419 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.420 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.421 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.422 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.424 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.425 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.426 - INFO:src.nhl_managers:[NHL] No recent games to display
Oct 01 21:12:58 ledpi ledmatrix[480]: 21:12:58.427 - INFO:src.nhl_managers:[NHL] No recent games to display

torontonightclub-oss, 2025-10-02:
this is the first part of my nhl_managers.py file is it not the latest one? I ran the update several times today

import os
import time
import logging
import requests
import json
from typing import Dict, Any, Optional, List
from PIL import Image, ImageDraw, ImageFont
from pathlib import Path
from datetime import datetime, timedelta, timezone
from src.display_manager import DisplayManager
from src.cache_manager import CacheManager
from src.config_manager import ConfigManager
from src.odds_manager import OddsManager
from src.background_data_service import get_background_service
import pytz

# Import the API counter function from web interface
try:
    from web_interface_v2 import increment_api_counter
except ImportError:
    # Fallback if web interface is not available
    def increment_api_counter(kind: str, count: int = 1):
        pass

# Constants
NHL_API_BASE_URL = "https://api-web.nhle.com/v1/schedule/"

# Configure logging to match main configuration
logging.basicConfig(
    level=logging.INFO,
    format='%(asctime)s.%(msecs)03d - %(levelname)s:%(name)s:%(message)s',
    datefmt='%Y-%m-%d %H:%M:%S'
)

class BaseNHLManager:
    """Base class for NHL managers with common functionality."""
    # Class variables for warning tracking
    _no_data_warning_logged = False
    _last_warning_time = 0
    _warning_cooldown = 60  # Only log warnings once per minute
    _shared_data = None
    _last_shared_update = 0

    def __init__(self, config: Dict[str, Any], display_manager: DisplayManager, cache_manager: CacheManager):
        self.display_manager = display_manager
        # Store reference to config instead of creating new ConfigManager
        self.config_manager = None  # Not used in this class
        self.config = config
        self.cache_manager = cache_manager
        self.odds_manager = OddsManager(self.cache_manager, None)
        self.logger = logging.getLogger(__name__)
        self.nhl_config = config.get("nhl_scoreboard", {})
        self.is_enabled = self.nhl_config.get("enabled", False)
        self.show_odds = self.nhl_config.get("show_odds", False)
        self.test_mode = self.nhl_config.get("test_mode", False)  # Use test_mode from config
        self.logo_dir = self.nhl_config.get("logo_dir", "assets/sports/nhl_logos")
        self.update_interval = self.nhl_config.get("update_interval_seconds", 60)
        self.show_records = self.nhl_config.get('show_records', False)
        self.last_update = 0
        self.current_game = None
        self.fonts = self._load_fonts()
        self.favorite_teams = self.nhl_config.get("favorite_teams", [])
        self.recent_hours = self.nhl_config.get("recent_game_hours", 48)  # Default 48 hours

        # Set logging level to DEBUG to see all messages
        self.logger.setLevel(logging.DEBUG)

legoguy1000, 2025-10-02:
> this is the first part of my nhl_managers.py file is it not the latest one? I ran the update several times today
>
> import os
> import time
> import logging
> import requests
> import json
> from typing import [...]

Ya definitely not up to date. How did u update? `git pull`? What branch are u on? `git branch`

torontonightclub-oss, 2025-10-02:
ill run git pull now and report back

torontonightclub-oss, 2025-10-02:
Unpacking objects: 100% (6/6), 19.61 KiB | 260.00 KiB/s, done.
From https://github.com/ChuckBuilds/LEDMatrix
   83f76db..e749925  main       -> origin/main
Updating 31ec001..e749925
error: Your local changes to the following files would be overwritten by merge:
        first_time_install.sh
        fix_assets_permissions.sh
        src/nhl_managers.py
Please commit your changes or stash them before you merge.
Aborting

closed 2025-10-02 by legoguy1000